Desuuuu/Marlin #1: Print Resume After Power-Failure Not Functioning

jlwainwright, 2020-06-21:
Printer: Creality CR10s-Pro V2

Firmware: https://github.com/Desuuuu/Marlin

System Board: Creality 3D v2.4

Issue: No Prompt To Resume Print After Restoring Power After Power Failure

Result Of M413 Echo:
"Power-loss recovery ON"

Desuuuu, 2020-06-21:
Did you have an SD card inserted? The print progress is saved on the SD card so it's required for PLR to function.

I don't think that's an issue upstream because PLR appears to work for some people on 2.0.5.3.

It might take me some time to track this issue down because I'm waiting for parts for an upgrade, so I can't test this myself at the moment.

jlwainwright, 2020-06-21:
The testing was done while printing from SD yes @Desuuuu 

Thank You.

Desuuuu, 2020-07-27:
The latest version based on Marlin 2.0.6 contains some fixes related to PLR.

I suggest giving it a try, it might have actually solved the issue.

closed 2020-09-24 by Desuuuu