Desuuuu/Marlin #6: [FR] Allow a custom size ABL grid

mrv96, 2020-11-22:
According to this line of code:

https://github.com/Desuuuu/Marlin/blob/1b614d3e4a0ae117b86b86dd30e2cfe3ac32baaa/Marlin/src/lcd/extui/lib/dgus_reloaded/config/DGUS_Constants.h#L28

ABL grid size must be composed exactly of 25 points. Why this limitation? Is there any way to remove it?

Desuuuu, 2020-11-22:
The two ABL screens ([1](https://raw.githubusercontent.com/Desuuuu/DGUS-reloaded/master/project/DWIN_SET/013_leveling_automatic.bmp) and [2](https://raw.githubusercontent.com/Desuuuu/DGUS-reloaded/master/project/DWIN_SET/014_leveling_probing.bmp)) expect a 5x5 grid. I don't think there's a way to have a proper UI with a variable grid size, hence why it's fixed.

It's possible to make it work with different grids regardless but the UI will be broken and that's something I don't really like.

mrv96, 2020-11-22:
Thank you for your reply. In order to have a huge grid the steps to performs are only:
- modify png
- modify DGUS LCD firmware
- remove the static_assert

Am i missing something?

Desuuuu, 2020-11-23:
That sounds about right.

In case you need new addresses for the screen, make sure you use new ones to avoid any issue.

mrv96, 2020-11-23:
Thank you. Since this feature request cannot be solved, I close the issue.

mrv96, 2020-11-26:
@Desuuuu what about loading circle instead of fixed on-screen grid during ABL procedure?

Desuuuu, 2020-11-26:
That's always an option, but both the probing & automatic leveling screens would need to be modified for it. The best way to do it would probably to make variants of these screens so the current functionality would be preserved for 5x5 grids.

I'm not going to be able to dedicate much time to this for at least a few month though.

I'll reopen this ticket anyway, as that would be a nice feature to have and there should be a proper way to implement it.