Desuuuu/Marlin #8: [FR] Support for AUTO_BED_LEVELING_UBL

mrv96, 2020-11-26:
Hi, as you already know i'm trying your firmware. I notice that AUTO_BED_LEVELING_UBL screen leveling is currently not supported. Could you add it? I would like to make a PR, but i guess that this task is much easier for you that already know the code.

Desuuuu, 2020-11-26:
Do you mean adding support to the touchscreen firmware for it? I'm not familiar about the leveling procedure with UBL.

mrv96, 2020-11-26:
Yes. I mean exactly that. UBL is not very different from the others, but has a different Gcode sequence  to be executed during the leveling procedure: https://marlinfw.org/docs/gcode/G029-ubl.html

The firmware could for instance performs exactly what is reported at the end of the previous url's page

closed 2021-01-09 by Desuuuu