Desuuuu/Marlin #10: No such file or directory[BUG] (short description)

akash1769, 2020-12-03:
For the CR10S Pro config when doing a verify I get this message: 

   avr-gcc: error: CreateProcess: No such file or directory

   exit status 1

   Error compiling for board Arduino Mega or Mega 2560.

the entry before it was quite verbose... didn't wanna muck up this message by putting it in. How would you like to have me present it?

Desuuuu, 2020-12-03:
This is most likely an issue with Arduino IDE and long file paths.

I highly suggest [compiling using PlatformIO](https://marlinfw.org/docs/basics/install_platformio.html) (in VSCode for example).

akash1769, 2020-12-04:
Thank you for that suggestion. I may not be at the level most people are using platformio. After a half hour of trial an error I managed to get the extension running in visual studio but not sure how to get past the tutorial on setting up a project. I duplicated what the tutorial suggested but not sure how to apply that to compiling Marlin. 

Docs.platformio.org under cloud&Desktop IDE > VSCode

akash1769, 2020-12-04:
Also I was successful in the past using Arduino IDE with the first version u created. What's different now in the current config? Haven't u been doing this all along in Marlin 2?

akash1769, 2020-12-04:
Re-Found the tutorial in Marlin for doing this. Had no idea the commitment I was signing up for 😆

Still curious why not Arduino IDE though.

Desuuuu, 2020-12-04:
AFAIK, there wasn't any recent change in my code that could cause this. This is probably an issue upstream, which you are experiencing now because Marlin was updated to version 2.0.7.2.

Anyway, I don't see much of a reason to use Arduino IDE with Marlin over VSCode.

VSCode is much more powerful and pleasant to use and you shouldn't experience bugs like this one with it.

closed 2020-12-16 by Desuuuu