Desuuuu/Marlin #11: [BUG] (error compiling MKS Robin E3D on Ender 5 Plus)

arsonists, 2020-12-17:
### Bug Description

 I have used the default configuration file that is published for Ender 5 Plus but with the settings to adapt it to the MKS Robin E3D.

At the time of compilation gives the following problem:

"Marlin\src\lcd\extui\lib\dgus_reloaded\DGUSDisplay.cpp: In static member function 'static void DGUSDisplay::EnableControl(DGUS_Screen, DGUSDisplay::DGUS_ControlType, DGUS_Control)':
Marlin\src\lcd\extui\lib\dgus_reloaded\DGUSDisplay.cpp:181:14: error: 'class MarlinSerial' has no member named 'flushTX'; did you mean 'flush'?
   LCD_SERIAL.flushTX();"

### My Configurations

[Marlin.zip](https://github.com/Desuuuu/Marlin/files/5709695/Marlin.zip)

Desuuuu, 2020-12-17:
The problem should be solved as of commit 52a94d8777171b25a65800cb69955fb7010bea7f. Let me know if you encounter any other issue.

arsonists, 2020-12-17:
Good night.

Thank you very much for your work, I have already managed to compile marlin.

Now I have the problem that after installing the firmware, the screen stays in the Marlin logo (of course I have updated with your firmware)...

I don't know what could be wrong...

Desuuuu, 2020-12-18:
Have you followed [these steps](https://github.com/Desuuuu/DGUS-reloaded/wiki/Flashing-the-firmware) to flash the touchscreen?

It could also be a communication issue between the mainboard and touchscreen. Make sure your `LCD_SERIAL_PORT` is correct.

closed 2021-01-09 by Desuuuu