Desuuuu/Marlin #15: [BUG] (Sd contents and folder name)

DaGr70, 2021-01-09:
### Bug Description

Hi,
I forked your version of Marlin include Dgus screen compatibilies  from :   [Desuuuu/Marlin](https://github.com/Desuuuu/Marlin) (Thanks for your great job)
[Thghca](https://github.com/thghca) pull my somes issues and I meet similar issues with  my version of firmware.

In resume :
- if Sdcard contain a number of files under 3 : the "selection" button don't react. (same issue when located into folders)
- if exist a folder name  with lenght  upper than 8 characters : the firmware don't reconize concerned  folder or another folder and buttons don't work properly
- If folder name contain "space" character => I meet similar issue

No issue with 4 files on sdcard and short folder names.

Thanks .

### My Configurations

[Configuration_files.zip](https://github.com/Desuuuu/Marlin/files/5790871/Configuration_files.zip)

### Steps to Reproduce

1.  put only one file on sd card
[or]
2. put one folder with long name (> 8 character)
[or]
3. put one folder with  name include "space" character

**Expected behavior:** [What you expect to happen]
Different icon for folder and file names
All file could be selected
All foderl could be opened

**Actual behavior:** [What actually happens]

Can't select file
folder don't be reconized by firmware

#### Additional Information
Can be added if necessay, but it seems easy to reach issue

Desuuuu, 2021-01-09:
I'm pretty sure the issue you're facing comes from your modifications and is irrelevant here.

Anyway, it's happening because in order to enable/disable controls in certain situations the firmware has to know on which screen they are and at which index they are stored in the screen firmware (this information can be found in the left panel of DGUS Tool).

If you've messed with how the controls are ordered, the firmware will disable the wrong control since it is selected based on its index.

You have two solutions:

* Make sure your screen firmware has its controls in the same order as mine.
* Update control indexes in [DGUS_Control.h](https://github.com/Desuuuu/Marlin/blob/2.0.x/Marlin/src/lcd/extui/lib/dgus_reloaded/config/DGUS_Control.h) for your firmware.

Regarding long folders, this is controlled by Marlin not by my code. You can try enabling `LONG_FILENAME_HOST_SUPPORT` to see if it makes a difference.

DaGr70, 2021-01-09:
Thanks Dessuuu for your feedback. 
In fact my screen firmware  index information was different. 
DGUS tool change them when I save projet with screen rotation.
I'll try your proposals to solve my issue.

Have a nice Day

closed 2021-01-09 by DaGr70