Desuuuu/Marlin #16: [BUG] Power Loss Recovery doesn't works

mrv96, 2021-01-09:
Hi Desuuu, I know that you merged your repo with Marlin 2.0.7.2 after a my request. Today I was trying the power loss feature, but it doesn't work. I found that unluckily the PLR feature is broken in Marlin 2.0.7.2, but it has been already fixed in the bugfix branch.

Related issue: https://github.com/MarlinFirmware/Marlin/issues/19602

I open this issue as reminder to merge next Marlin release in your repo asap.

P.S: a PR of your excellent work in official Marlin would avoid this :)

Astador, 2021-02-27:
Hi @mrv96 Any luck with this?? i'm trying to use PLR feature but get stuck after reach last point printed. did you resolve the issue?

mrv96, 2021-02-28:
Hi, here the list of possible solutions:

- Wait for next Marlin release
- Substitute `powerloss.cpp` and `powerloss.cpp` of this repo with the ones in the current official Marlin bugfix (not tested)
- Substitute `powerloss.cpp` and `powerloss.cpp` of this repo with the ones of Marlin 2.0.6.1 (working)
- Apply this https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/444#issuecomment-748154694 (working)

Desuuuu, 2021-02-28:
The [bugfix branch](https://github.com/Desuuuu/Marlin/tree/bugfix) now has my modifications applied to Marlin's `bugfix-2.0.x` branch, so it should contain the fix for this issue.

As a side note, I also removed the 2.0.7.2 release as it was causing too many issues.

mrv96, 2021-04-30:
Marlin 2.0.8 has been just released: https://github.com/MarlinFirmware/Marlin/releases/tag/2.0.8

closed 2021-08-22 by Desuuuu