Desuuuu/Marlin #24: Spaces in GCode input and steppers not off when print aborted.

DaKrampus, 2021-02-28:
First of all Desuuuu, i want to stress that this firmware works like a charm on a CR10Spro. I had lots of problems with the TM firmware, but this one is great. There are 2 features I would love to see though, but i do not know how easy/difficult they are to implement.

1. I know that spaces are not necessary in Gcode. But if you have a code with 4 parameters you want to input, it is so much easier if you can input a space between the parameters. but i agree it works without it. 

2. It certainly not a bug, because TM also does it that way, even at the end of a print. Head parks with steppers OFF, but when you click (press) done the steppers are ON again, which i find annoying, because you want to pul the plate towards you and maybe clean it, for the next print...
In your version it is perfect, because the steppers remain off like it is expected by me.

**But if you abort a print**, the nozzle parks itself and the steppers **remain ON**. To be able to move the build plate etc... i have to stop them manually. I can live with that. but it would be nice if they behaved like at the end of a print. ==> steppers OFF.

Thats my 2 cents.
But if i may, i think the overall utility of this firmware is far superior the the Stock AND to the TM one. 
I am really happy with this one, and  must say it gives me a much better feeling when printing. I have the impression i have a better control over my printer with this firmware.
Thanks again for your great work

Luc

### Description

### Feature Workflow

1. [First Action]
2. [Second Action]
3. [and so on...]

#### Additional Information

* Provide pictures or links that demonstrate a similar feature or concept.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Desuuuu, 2021-02-28:
> I know that spaces are not necessary in Gcode. But if you have a code with 4 parameters you want to input, it is so much easier if you can input a space between the parameters. but i agree it works without it.

I already plan on remaking the keyboard. Thing is, I also plan on completely refactoring the firmware (into something more flexible), so I'm not sure I wanna invest time into updating the keyboard before that.

> if you ABORT a print, the nozzle parks itself and the steppers remain ON

That's the expected behaviour with the default configuration.

You can change with the `EVENT_GCODE_SD_ABORT` option in `Configuration_adv.h` if you want to change this.

DaKrampus, 2021-02-28:
> You can change the `EVENT_GCODE_SD_ABORT` option in `Configuration_adv.h` if you want to change this.

Thank you i will .. great!
Dont worry for the keyboard, it is not something you use very often.. (at least I dont) just once in while...
thanx again.

Luc

closed 2021-03-09 by Desuuuu