Desuuuu/Marlin #25: [BUG] Issue with filament detection

DaKrampus, 2021-03-28:
I did have a broken filament on a CR10S Pro and it did stop, although it was a bit late.
but i had another problem with this one and a second cr10s pro.
I wanted to start printing and had forgotten to load the filament. i realized it quite quickly, but Instead of stopping the print and putting in the filament, i let the printer continue to see what it would do... well it started printing although the stock filament detector was off (the light only lights when a filament is present).
In my understandig it should not start to print.

I tried to reproduce  the error with the other printer and it showed exactly the same behaviour.
i.e. starts to print without filament.
I know you dont have time for this..
but any hint to where to look in the firmware?

I am using your bugfix 2.0x version..
any files i should replace from the original marlin bugfix version ?

Luc

Desuuuu, 2021-03-28:
Can you check what's reported by `M119` with and without the sensor triggered?

Unless you have a specific need to run a more recent version, I highly suggest you try version 2.0.6.1 as well.

DaKrampus, 2021-03-29:
Thank you for the quick response.. 
i did what you told me and saw that it does work. although it did some confusing things for me. But the log file speaks for itself.

But first of all: The reason I am using the bugfix version is your fix for the pronterface access.
A month ago, you wrote that you had added the fix in the bugfix version. that is why i am using it.
and it is working great.

Now thank you for your hints, it resolved my problem.

I should have read the config file first:
Quote:
_* IMPORTANT: Runout will only trigger if Marlin is aware that a print job is running.
 * Marlin knows a print job is running when:
 *  1. Running a print job from media started with M24.
 *  2. The Print Job Timer has been started with M75.
 *  3. The heaters were turned on and PRINTJOB_TIMER_AUTOSTART is enabled._

**That is why it had started printing before giving alert.!!!**
In future i will read the details before posting.. promised.

I did what you said and it shows every thing ok...

The only strange thing is, (it does it sometimes) at the beginning of the print it asks me if i want to resume. if I say yes it then continues.
But it does not tell me that something is wrong.. just requester: resume or not (see the logfile below)
Usually it starts printing directly without that resume question... 

after i tell it to resume it starts heating and prints the left line, and only after having printed that, gives the Filament runout alert.

That is why i thought it had started really printing and switched the printer off before i could see the alert.
The other printer behaves exactly the same. Also here i switched off the machine so it would not continue after having printed the left "priming" line without filament.

I am sorry for having alerted you with a bug that is not one.
Thank you again for your great work.
Luc

Here ist the proterface log.

// ************* copy paste start **********//

Connecting...
start
Printer is now online.
 External Reset
Marlin bugfix-2.0.x
echo: Last Updated: 2021-02-27 | Author: Desuuuu
echo:Compiled: Mar 10 2021
echo: Free Memory: 3027  PlannerBufferBytes: 1472
//action:notification CR-10S Pro Ready.
//action:prompt_end
echo:SD card ok

//*** With Filament
>>> m119
SENDING:M119
Reporting endstop status
x_min: TRIGGERED
y_min: open
z_min: TRIGGERED
filament: TRIGGERED
ok P15 B3

//*** I quickly did a home

echo:enqueueing "G28XYZ"

//*** then a m119 again with filament

>>> m119
SENDING:M119
Reporting endstop status
x_min: open
y_min: open
z_min: TRIGGERED
filament: TRIGGERED
ok P15 B3

//*** then a m412

>>> M412
SENDING:M412
echo:Filament runout ON
Filament runout distance (mm): 20.00
ok P15 B3

//*** removed the filament

SENDING:M119
Reporting endstop status
x_min: open
y_min: open
z_min: TRIGGERED
filament: open
ok P15 B3

// Now i start a print with NO filament.. //

echo:SD card ok
echo:Now fresh file: xx_sd-~1.gco
File opened: xx_sd-~1.gco Size: 3962319
File selected
//action:notification XX_SD-CardBox_v1.0_R.gcode
ok P15 B4
//action:resume
//action:prompt_end
//action:prompt_begin Resuming SD
//action:prompt_button Dismiss
//action:prompt_show
//action:notification XX_SD-CardBox_v1.0_R.gcode
ok P15 B4
//action:notification Time Left 5h17m26s
//action:notification Bed Heating...

//******* nozzle is not heating as i set the temp manually to 200 for faster operation

//action:notification XX_SD-CardBox_v1.0_R.gcode
//action:notification XX_SD-CardBox_v1.0_R.gcode
//action:prompt_end
//action:prompt_begin M1 Stop
//action:prompt_button Continue
//action:prompt_show
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
ok P15 B0
echo:Bed Leveling ON
echo:Fade Height 5.00
Writing to file: /PLR

//*** About here he has allready printed the starting line on the left side.. (for me it meant it has started real printing!!!)
//*** I stopped it manually before it could go any further -  that is why i did not see the filament runout requester that comes only later)

//action:prompt_end
//action:prompt_begin FilamentRunout T0
//action:prompt_show
//action:paused filament_runout 0
//action:paused
echo:Insert filament and send M108
//action:prompt_end
//action:prompt_begin Nozzle Parked
//action:prompt_button Continue
//action:prompt_show
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
//action:cancel
//action:notification Print Aborted
//action:prompt_end
//action:prompt_begin Paused
//action:prompt_button PurgeMore
//action:prompt_button DisableRunout
//action:prompt_show
//action:prompt_end
//action:notification Print Paused
Writing to file: /PLR
//action:resumed
//action:notification XX_SD-CardBox_v1.0_R.gcode
File deleted:PLR
ok P15 B0
ok P14 B0

// ************* copy paste end **********//

Again sorry for your time. 

Luc

FrancYescO, 2021-03-29:
check if this is the culprit: https://github.com/DaGr70/Marlin_Longer_LK4_pro/commit/3e318f52fa687610aedf3d836be80517c5842d32#diff-bb298d18dac4268225a7a53a5ad24844472530bdc63f6a98a2ba8d8158b75080R1207

Desuuuu, 2021-03-29:
Version 2.0.6.1 is not affected by this serial issue.

I'm pretty sure the resume prompt you see at the beginning of your prints is because you have an M1 command in your G-code.

Desuuuu, 2021-03-29:
`FILAMENT_RUNOUT_DISTANCE_MM` is set to 20mm in the default configurations.

It might explain why the firmware prints for a bit before pausing when you start a print without filament.

DaKrampus, 2021-03-29:
Yes.. you can close the bug.. everything works a designed.

Yes you are right, it first uses the 20 mm.. so it starts printing. then.. as the filament alert is given it runs M600

which my printer doesnt like at all.. it first of all unloads the filament.. which makes the extruder run backwards for 400 mm (!) or so.. that is when i says "busy" and i cannot do anything!!! ( i did not realize that)
of course if i dont have any filament inside i can hardly see the extruder pulling out the filament...

then there is a resume button and again.. it will say "busy" because it is loading like 400 mm of filament up through the extruder.. before it primes ... then primes a couple of times... 
AND
in between it stops somewhere and sends itself a M1 because, while it was loading,  i pushed like 100 times the resume or continue button.. it was saying busy all the time, but still accepted the resume.. and somehow it did not like it. so it did a full stop. Its not a bug.. i just wasnt aware that it was doing something, something i did not know it was doing.

I think that by changing FILAMENT_RUNOUT_SCRIPT from M600 to something else...
like M25 and turn off heating... i would be on the safe side if it runs out of filament while i am not at home or sleeping.

So yes you can close the bug.. because it is not one!

Thanks for all the help and your time.

Luc

closed 2021-03-29 by Desuuuu