Desuuuu/Marlin #30: [BUG] On Geeetech Printers (A10T, A20, A20M using 2.0.9.1 the Y axis will not home properly.

andraisr, 2021-09-10:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

I recently started testing out 2.0.9.1. I am finding the when homing the Y axis, halfway to the home position there is a slight hesitation, then I get a Homing failure msg on the LCD. This is on several Geeetech Printer models (see title line). I revert back to 2.0.1 and when I home the Y Axis, it is smooth and homes properly. This happens with Octoprint and when directly connected to the computer via USB cable using S3D. 

This issue is duplicatable on all the models mentioned. If the bed is close to the home position, it will work. But further than 1/2 way, if fails everytime.

### Bug Timeline

This is a new bug for me. Not sure if this has been reported before.

### Expected behavior

I expect home to work with the y axis.

### Actual behavior

The movement of the bed, hesitates, then error message on LCD indicationing a Homing Failure. 

### Steps to Reproduce

1. Move the Bed more than half its distance.
2. Then Octoprint or S3D to home Y axis.
3. Error occurs.

### Version of Marlin Firmware

2.0.9.1

### Printer model

Geeetech A10T, A20, A20M

### Electronics

Stock

### Your Slicer

Simplify3D

### Host Software

OctoPrint

### Additional information & file uploads

I repair these printers for Geeetech. I get them to stock with no hardware upgrades.

Desuuuu, 2021-09-10:
Unless I'm mistaken, these printers are not equipped with a DGUS touchscreen so I doubt your issue has anything to do with this fork. Please direct your issue upstream.

andraisr, 2021-09-10:
Please advise on how to upstream this issue? like, who do I send it to?

Thanks,

Israel

Desuuuu, 2021-09-10:
The official Marlin repository: https://github.com/MarlinFirmware/Marlin

closed 2021-09-10 by Desuuuu